clssn/numato-gpio #2: Support for 16 port and 8 channel relay

docsparks, 2020-05-24:
I have a 16 port usb gpio expander from numatolabs and I know because of the use of the hexdecimal conversion for port #s that the code will error with my 16port vs the 32.

I also have a 8 channel usb relay from numatolabs and would like to help contribute to a release for that also. 

im not experienced with github yet, but anything I can do to help contribute, please let me know.

Will post errors when i have a moment to test hassio intergration with my board.

clssn, 2020-07-21:
Sorry, actually I missed the creation of this issue. It's my first open sourced piece of software. So I really appreciate your interest. 
If you have some Python knowledge you may be able to figure it out yourself. The GitHub way would be that you fork my repo, clone your fork, prepare the changes on a feature branch on your machine, push the branch to your fork and pull-request it back to me. I'd then review your contribution and request/propose changes and so on.

So first this python module and in case you're a HomeAssistant user, we could then update the HA numato component and related documentation as well.

Just in case you're still interested.

clssn, 2020-08-02:
Hm, if you're still interested, @docsparks, I could try and make the changes and you run the sys_tests with `pytest sys_tests` with your connected boards and give me feedback. Just doesn't make sense for me to implement this without having the chance to test it.

clssn, 2020-08-02:
@docsparks, You may want to try https://github.com/clssn/numato-gpio/tree/support-all-numato-usb-gpio-devices
I might ask the Numato guys for a test of all devices as well.

docsparks, 2020-08-05:
I will get back to you in the next day or so, thank you again for your efforts.

clssn, 2020-11-15:
Hi @docsparks, any chance, you test my branch?

clssn, 2020-12-06:
I'll close this issue due to inactivity. Feel free to re-open or open a new issue if you're still interested.

closed 2020-12-06 by clssn