clssn/numato-gpio #3: Mapped Ports don't show as Entities

HugoFJ, 2020-07-05:
I can't get mapped ports (sensors, switches or binary sensors) to create entities within Hassio OS 4.11 
I'm using RPi4
For a sanity check I've also tried a fresh Hassio install with no other integrations and copied the numato example .yaml into the configuration.yaml as below.
numato: 
  devices:
    - id: 0
      binary_sensors:
        ports:
          2: Window Livingroom Open
          3: Window Livingroom Glassbreak
          4: Doorbell
      sensors:
        ports:
          1:
            name: Soil Moisture Ficus
      switches:
        ports:
          5: Relay Light Outdoor
          6: Relay Circulation Pump
          7: Door Opener

This is doesn't create any entities either, am I missing something or is it broken? 
Many thanks, Hugo

clssn, 2020-07-15:
Hi, sorry for the delay. I need to adjust my mail filter. Unfortunately I'm not using Hass OS. Just running a docker container (with HA v0.112.4).

1) Do you have a Numato 32 Port USB GPIO board? 
2) When you connect it to your Raspi, will it be assigned to some /dev/ttyACMx device? You can check that with `dmesg` right after connecting.
3) I guess you haven't changed the board's id have you?

You can test whether your board can be found using my python tool from https://github.com/clssn/numato-gpio.
Install it as described in the readme, then just run `python3 -m numato_gpio`. Since the ACM devices are made for modems they belong to the dialout group, so your user needs to be member of the dialout group. Alternatively use root: `sudo python3 -m numato_gpio` but be careful to not trust a stranger with his python code and have a look at the code first :-)

Expect some output like:
```
dev: /dev/ttyACM0 | id: 0 | ver: 9 | iodir: 0xffffffff | iomask: 0x00000000 | state: 0x00000000
dev: /dev/ttyACM1 | id: 1 | ver: 9 | iodir: 0xffffffff | iomask: 0x00000000 | state: 0x00000000
```

Small hint:
```
When you paste a code block like the yaml config,
    enclose it in a pair of triple backticks (markdown syntax).
    This will keep the formatting clean like in this block and help identify issues.
```

HugoFJ, 2020-07-17:
1. Yes I do have the Numato 32 Port GPIO board
2. I've used the "Terminal & SSH" add on from supervisor. Using the `ha hardware info` command I can see that the board is connect as `/dev/ttyACM0`
3. I tried changing it (with my windows machine) to see if it made any difference. I can control the board using Putty in windows and I reset the ID to 0

It seems a bit of pain to SSH into Hassio, I'm seriously considering porting everything over and running HA in a docker on the RPi4, is there any disadvantage to doing this?

clssn, 2020-07-17:
Well, as far as I understand the Hassio OS fully manages the Home Assistant installation and operation including update via UI, restart after a crash or power down. Maybe even some backup of your config? While with docker it's more manual work. While you can tell docker to restart the container whenever it was stopped also after reboot, you'll need to bump the container version manually and have docker recreate the container. With a docker compose file that's easy though. Also you need to take care of your config.

My guess for your problem would be that in Hassio OS the user running Home Assistant does not belong to the dialout group. So you'll need console access and superuser privileges to change that.

HugoFJ, 2020-07-17:
Any clues as to how to get the Hassio OS user to belong to the dial out group? To be honest I've no idea how to achieve console access and have super user privileges.

clssn, 2020-07-21:
Sorry @HugoFJ, since I have no Hassio OS I don't have a clue. Besides this is probably a more general issue other users might have as well. So better try to get the info via the [Home Assistant Community](https://www.home-assistant.io/getting-started/join-the-community/). I'm interested in the solution though, in order to augment the documentation for other Hassio OS users with Numato device (you could contribute that as well once you found out).

clssn, 2020-07-21:
Not sure whether that's still the way, but that's what I found: https://community.home-assistant.io/t/how-do-i-access-hass-io-via-ssh/83561/2
With ssh root access you'd need to
1) check whether the device /dev/ttyACM0 is added when you connect the Numato board and look which group it belongs to (let's assume `dialout`)
2) Determine the username of the user running home-assistant: Call `ps -ef` look for a line like `python3 -m homeassistant --config /config` and take the username from the second column, let's assume it's `username`
3) run `adduser username dialout` to add the user to the group.

clssn, 2020-08-02:
@HugoFJ, did this work for you?

HugoFJ, 2020-08-03:
> @HugoFJ, did this work for you?

Sadly not really. Calling `ps -ef` didn't reveal anything that looked like `python3 -m homeassistant --config /config`

I did try googling to find out what user name Hassio OS might be using but with no luck. The command `add root dialout` works, I had a guess at what other user names to use and would get a error that they were wrong...

I've tried the forum but there's no reply to any of my posts!

I found this for enabling I2C https://www.home-assistant.io/hassio/enable_i2c I wonder if there's anything to add in the config text to enable the dialout group?

(You can tell I'm well beyond the limits of my knowledge now!)

clssn, 2020-08-07:
Don't worry, I know what you're talking about. Home Assistant is developing so quick. I sometimes thought I'd understand something and the next day it was changed or gone at all. If you can preserve a positive attitude about that agility (which I happen to do most of the time) I guess it keeps you fresh and flexible ;-)
Regarding your problem: I plan on setting up a Raspi with Hassio for my parents' home soon. Though they don't need a Numato board, I'd try to get my spare one working and document how I did it. So hang on.
Best,
Henning

HugoFJ, 2020-08-07:
Amazing, thanks Henning :-) I'll hold tight and await you findings.

Thanks for developing and publishing this integration, it's greatly appreciated.

Hugo

clssn, 2020-08-14:
Hey @HugoFJ, I installed Hassio on a Raspi3. Connected my Numato GPIO 32 and it showed up under Supervisor / System / Host System / Hardware as `/dev/ttyACM0`. I pasted the configuration example from the numato documentation into `configuration.yaml` and after a restart of Home Assistant it worked right away. It's that straight forward that I wouldn't document it.

For writing the `configuration.yaml` there are two major ways I found. I installed the ssh addon (it is offered in the addon store once you select _Advanced Mode_ in your user profile page. Opening the terminal you can use `vi` or `nano` editors like `nano config/configuration.yaml` and paste the config e.g. with `Ctrl+Shift+v`.
Another approach would be the samba addon. It should offer the config directory as a network share (especially useful if you're a windows guy and don't like the command line :-)

Entities show up in the entities view as well for me.

I suggest you first have a look at the "hardware info" in the Supervisor UI and look for a `/dev/ttyACM0`. If you have that and things still don't work, please attach the device on any Linux machine with Python 3 installed and try whether the numato_gpio module can discover the device: `pip3 install numato_gpio` and then `python3 -m numato_gpio`. This should also show whether the device ID is the one you expected. If the device is accessible via `/dev/ttyACM0` but `numato_gpio` can't discover it, your device may have a different firmware version (though the numato documentation says nothing about differences of firmware versions I think). Mine has version 9 as you can see in this repository's readme. 

One further though: Do you have a recent version of Hassio?

Curious how you get along. Best, Henning

clssn, 2020-08-14:
Closed by mistake. Sorry.

clssn, 2020-11-15:
@HugoFJ , just out of curiosity, have you had any luck getting your Numato device running on HassIO?

clssn, 2021-01-18:
Closing the ticket due to inactivity.

closed 2021-01-18 by clssn