clssn/numato-gpio #9: undiscovered device

Froma, 2021-01-16:
I've got a numato 32 USB GPIO, the latest version of numato-gpio has been installed with all the requirements. 
When I run the command: `python3 -m numato_gpio`

the answer is: `Discovered devices: (None)`

Numato-gpio version: 0.9.0.dev0
O.S version: Ubuntu 20.04
Python version: 3.8.2
Libraries:
```
pyserial 3.5
pytest 5.4.3
pytest-timeout 1.4.2
tox 3.18.1
```

The O.S. it is installed on the Olimex A20-OLinuXino-LIME2

Using the python Serial libraries or with the screen linux command everything works fine.

Thanks!

clssn, 2021-01-16:
Hi, first of all, is your device available as `/dev/ACM0` (or another number) when you attach it via USB? If so, please check the file permissions for example with `ls -l /dev/ttyACM0`. For me it belongs to root and the dialout group. If it's the same for you, please make sure that your user is member of the dialout group or whatever group the device belongs to. For example with the command `sudo adduser username dialout`.

Hope that already helps..
Best regards
Henning

clssn, 2021-01-16:
Please check out my brand new troubleshooting document (linked in the README) and maybe use the troubleshooting script as explained, to test the exact device responses.

Froma, 2021-01-18:
Hi Henning,
thanks for your time!
I'm following all your advice, everthing seems OK, except for dmesg output

```
 olimex@pds2:~/Froma$  dmesg
[ 2222.163355] usb 4-1: new full-speed USB device number 7 using ohci-platform
[ 2222.398561] usb 4-1: New USB device found, idVendor=2a19, idProduct=0802, bcdDevice= 1.00
[ 2222.398593] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2222.398614] usb 4-1: Product: CDC RS-232 Emulation Demo
[ 2222.398634] usb 4-1: Manufacturer: Microchip Technology Inc.
[ 2222.410889] cdc_acm 4-1:1.0: ttyACM0: USB ACM device
 olimex@pds2:~/Froma$
```
 
I miss two lines respect your output
 **[719716.215317] usbcore: registered new interface driver cdc_acm
 [719716.215318] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters**
 
Do you think it could be a problem?
For everything else there is no difference...

Froma, 2021-01-18:
I tried on my raspberry, the `dmesg `output is like yours

```
[  162.755567] usb 1-1.2: new full-speed USB device number 4 using dwc_otg
[  162.897795] usb 1-1.2: New USB device found, idVendor=2a19, idProduct=0802, bcdDevice= 1.00
[  162.897815] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  162.897829] usb 1-1.2: Product: CDC RS-232 Emulation Demo
[  162.897843] usb 1-1.2: Manufacturer: Microchip Technology Inc.
[  162.958091] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[  162.959062] usbcore: registered new interface driver cdc_acm
[  162.959072] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

On `screen`, I've seen that this (as on olimex):
```
>gpio readall
DFFFFFFF
```

The result is the same I get on olimex
```
pi@raspberrypi:~ $ python3 -m numato_gpio
Discovered devices: (None)
[waiting...]
```

clssn, 2021-01-18:
Now that's unexpected: Your seem to use the same device with the same firmware version as I do. Yet the line endings returned by your device are '\n\r' while in my device they are '\r\n'.
I'll look into that asap and try to find a good solution to support both. So hang in there ;-)

clssn, 2021-01-19:
Can you please test whether my branch named `tolerate-different-eol-sequences` works for you?
In case you're not familiar with git and pip it's like:

    pip3 uninstall numato-gpio
    git clone -b tolerate-different-eol-sequences https://github.com/clssn/numato-gpio.git
    pip3 install --user ./numato-gpio
    python3 -m numato_gpio
    python3 -m numato_gpio.troubleshoot

Looking forward for your feedback!

Froma, 2021-01-20:
Everything seems to be working fine!

These are my results:
```
olimex@pds1:~$ python3 -m numato_gpio
Discovered devices:
dev: /dev/ttyACM0 | id: 0 | ver: 9 | iodir: 0xffffffff | iomask: 0x00000000 | state: 0x00000000

```
and
```

olimex@pds1:~$ python3 -m numato_gpio.troubleshoot /dev/ttyACM0
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'gpio notify off\r'
response (40 byte):
b'gpio notify off\n\rgpio notify disabled\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x6f 0x66 0x66 0xa 0xd 0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x64 0x69 0x73 0x61 0x62 0x6c 0x65 0x64 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\r00000009\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x39 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ffffffff\r'
response (23 byte):
b'gpio iomask ffffffff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00000000\r'
response (23 byte):
b'gpio iomask 00000000\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (25 byte):
b'gpio readall\n\rFBFFFFFD\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x42 0x46 0x46 0x46 0x46 0x46 0x44 0xa 0xd 0x3e
--

```

Do you need more information?
Thank you again!

clssn, 2021-01-20:
Great, thanks for the feedback. I'll prepare a release then. A little later I'll add support for the other devices, which I fixed and tested (for an 8 port device) in issue #7. 

May I ask you to try the branch `support-all-numato-usb-gpio-devices`? If it works for you that would be already one more device variant tested. Besides, it would be a pity if the release after the next already broke your setup again.. :-O

clssn, 2021-01-20:
FYI, I just released version 0.9.0 which includes the changes of the branch `tolerate-different-eol-sequences`. I'll delete that feature branch now.

Froma, 2021-01-21:
I've tested your branch on NUMATO USBGPIO8, NUMATO USBGPIO16 and NUMATO USBGPIO32.
These are my result:
**8PORT**
```
olimex@pds2:~/numato-gpio$ python3 -m numato_gpio
Discovered devices:
dev: /dev/ttyACM0 | id: 0 | ver: 8 | ports: 8 | iodir: 0xff | iomask: 0x00 | state: 0x00
```
```
olimex@pds2:~/numato-gpio$ python3 -m numato_gpio.troubleshoot /dev/ttyACM0 8
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\r00000008\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x38 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ff\r'
response (17 byte):
b'gpio iomask ff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00\r'
response (17 byte):
b'gpio iomask 00\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (19 byte):
b'gpio readall\n\r00\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x30 0x30 0xa 0xd 0x3e
--

```

**16PORT**
```

olimex@pds2:~$ python3 -m numato_gpio
Discovered devices:
dev: /dev/ttyACM0 | id: 0 | ver: 9 | ports: 16 | iodir: 0xffff | iomask: 0x0000 | state: 0x0000

```

```
olimex@pds2:~$ python3 -m numato_gpio.troubleshoot /dev/ttyACM0 16
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\r00000009\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x39 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ffff\r'
response (19 byte):
b'gpio iomask ffff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 0000\r'
response (19 byte):
b'gpio iomask 0000\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (21 byte):
b'gpio readall\n\rFFFF\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x46 0x46 0x46 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio notify off\r'
response (40 byte):
b'gpio notify off\n\rgpio notify disabled\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x6f 0x66 0x66 0xa 0xd 0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x64 0x69 0x73 0x61 0x62 0x6c 0x65 0x64 0xa 0xd 0x3e
--

```

**32PORT**

```
olimex@pds1:~$ python3 -m numato_gpio
Discovered devices:
dev: /dev/ttyACM0 | id: 0 | ver: 9 | ports: 32 | iodir: 0xffffffff | iomask: 0x00000000 | state: 0x00000000
```

```
olimex@pds1:~$ python3 -m numato_gpio.troubleshoot /dev/ttyACM0 32
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\r00000009\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x39 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ffffffff\r'
response (23 byte):
b'gpio iomask ffffffff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00000000\r'
response (23 byte):
b'gpio iomask 00000000\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (25 byte):
b'gpio readall\n\rFBFFFFFD\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x42 0x46 0x46 0x46 0x46 0x46 0x44 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio notify off\r'
response (40 byte):
b'gpio notify off\n\rgpio notify disabled\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x6f 0x66 0x66 0xa 0xd 0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x64 0x69 0x73 0x61 0x62 0x6c 0x65 0x64 0xa 0xd 0x3e
--
```

I hope I have been helpful.
Thanks!

clssn, 2021-01-21:
Wow, that looks great. 
I'll try to add unit tests for notifications, try it with my home assistant installation and then I'll probably release it.
Thanks for taking the time! I'll close the issue now. Don't hesitate to re-open or open a new issue in case you still got something.
Best regards
Henning

closed 2021-01-21 by clssn