clssn/numato-gpio #11: OSError: [Errno 9] Bad file descriptor

gaborszalma, 2021-01-24:
Hi! I just installed a fresh debian10 system, then 'pip3 install numato-gpio'
Connected my Numato USB GPIO 32, then run the command below. Am I doing something wrong, or it's a bug?
Thanks!!

root@debian:/home/user# python3 -m numato_gpio
Discovered devices:
dev: /dev/ttyACM0 | id: 0 | ver: 9 | iodir: 0xffffffff | iomask: 0x00000000 | state: 0x00000000
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/numato_gpio/__init__.py", line 426, in _poll
    current_value, previous_value, buf = read_notification()
  File "/usr/local/lib/python3.7/dist-packages/numato_gpio/__init__.py", line 403, in read_notification
    buf = self._ser.read(1).decode().replace("\r", "\n")
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/numato_gpio/__init__.py", line 450, in _poll
    self._ser.close()
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 533, in close
    os.close(self.fd)
OSError: [Errno 9] Bad file descriptor

clssn, 2021-01-25:
Hey @gaborszalma,
thanks for reporting the issue. That sounds as if the device had been closed by something or someone close to the end of the program (since your output looks fine apart from the exceptions at the end).

Some questions to rule some other possibilities out:
- Some issue reported by the kernel log (run `dmesg` after this happens)?
- Some other process using the device (check with `lsof | grep ttyACM0`)?
- Does this happen everytime you run the numato_gpio module?

Maybe run the troubleshooting script (refer to the [troubleshooting documentation](https://github.com/clssn/numato-gpio#troubleshooting)) and post your device's response.

In the last release (0.9.0) I changed the polling-thread to be a daemon thread, to avoid errors terminating the polling thread. This change could be a source for trouble. For me as well as two other users I was in contact about this, it worked. But here it looks like all threads complain about the closed serial line during 'shutdown' or destruction.

gaborszalma, 2021-01-25:
Hi!

One additional information, I run it in a Virtual Machine. When I plug in the device, Vmware asks me if I want to plug it into Linux or Mac. It's visible in /dev and screen command works under linux.

When I plug in the device, the output of dmesg:

[Mon Jan 25 16:39:58 2021] usb 2-2.2: new full-speed USB device number 6 using uhci_hcd
[Mon Jan 25 16:39:58 2021] usb 2-2.2: New USB device found, idVendor=2a19, idProduct=0802, bcdDevice= 1.00
[Mon Jan 25 16:39:58 2021] usb 2-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Mon Jan 25 16:39:58 2021] usb 2-2.2: Product: CDC RS-232 Emulation Demo
[Mon Jan 25 16:39:58 2021] usb 2-2.2: Manufacturer: Microchip Technology Inc.
[Mon Jan 25 16:39:58 2021] cdc_acm 2-2.2:1.0: ttyACM0: USB ACM device

There is no new dmesg message when I run **python3 -m numato_gpio**

There is no output when I run **lsof | grep ttyACM0**

Yes, it happens everytime I run **python3 -m numato_gpio**

The output of the troubleshooting script:

root@debian:/home/user# python3 -m numato_gpio.troubleshoot /dev/ttyACM0
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'gpio notify off\r'
response (40 byte):
b'gpio notify off\n\rgpio notify disabled\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x6f 0x66 0x66 0xa 0xd 0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x64 0x69 0x73 0x61 0x62 0x6c 0x65 0x64 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\r00000009\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x39 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ffffffff\r'
response (23 byte):
b'gpio iomask ffffffff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00000000\r'
response (23 byte):
b'gpio iomask 00000000\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (25 byte):
b'gpio readall\n\rFEEFFFFF\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x45 0x45 0x46 0x46 0x46 0x46 0x46 0xa 0xd 0x3e
--

Thank you!

gaborszalma, 2021-01-25:
Sure thing! Just let me know what to do

clssn, 2021-01-26:
First thing, please install the branch `support-all-numato-usb-gpio-devices` which is a refactoring and generalization plus a couple of bug fixes. Not all unit tests pass so far, but that shouldn't be a problem in your case.
If you don't know how to install the branch try: `pip install --upgrade --user git+https://github.com/clssn/numato-gpio.git@support-all-numato-usb-gpio-devices#egg=numato-gpio`

If that doesn't help, we'll try the next thing.

clssn, 2021-02-12:
Hi @gaborszalma !
I installed a Ubuntu on VMWare and could reproduce your problem, with numato-gpio v0.9.0.
Good news: I just released v0.10.0 and it appears to fix your problem as far as I can tell.
I'd be happy if you could confirm it's working for you as well.

gaborszalma, 2021-03-04:
Yes, it works with debian too! Sorry for the late reply. 
Thanks!

clssn, 2021-03-06:
Thanks for the reply, happy to help!

closed 2021-03-06 by clssn