clssn/numato-gpio #13: Library Doesn't Recognize My Numato 8 Port GPIO

CraigerCDT, 2021-11-26:
I'm trying to get my shiny new numato 8 port gpio working with python, and having no luck.
I loaded this library with pip, and it appears to be loaded fine.
I installed my gpio and loaded the drivers from numato, and it works fine using putty (COM6).
When I perform the **python -m numato_gpio** command to verify communication with the board, I receive a **Discovered devices: (None)** response. 
I tried actually doing some examply pythonic commands, and get a fail message when I do a dev=gpio.devices[my_device_id] that states 'KeyError: 0'..
Any ideas what may be going on?
Thanks!

CraigerCDT, 2021-11-26:
Just noticed the troubleshooting link.
Also, forgot to mention I'm on windows, and someone newish to Python, although I've programmed other languages.
I ran the python -m numato-gpio.troubleshooter command, and received the following:

Testing device COM6 with pyserial==3.5

writing to COM6: b'id get\r'
response (19 byte):
b'id get\n\r12345678\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38 0xa 0xd 0x3e
--
writing to COM6: b'ver\r'
response (16 byte):
b'ver\n\r00000009\n\r>'
0x76 0x65 0x72 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x39 0xa 0xd 0x3e
--
writing to COM6: b'gpio iomask ff\r'
response (17 byte):
b'gpio iomask ff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0xa 0xd 0x3e
--
writing to COM6: b'gpio iomask 00\r'
response (17 byte):
b'gpio iomask 00\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0xa 0xd 0x3e
--
writing to COM6: b'gpio readall\r'
response (19 byte):
b'gpio readall\n\r00\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x30 0x30 0xa 0xd 0x3e
--

CraigerCDT, 2021-11-26:
Ok, did some digging in your code after installing the lib, changed the *nix device prefix to 'COM' and left the rest alone.
Now when I do the python -m numato_gpio command I get:
C:\>python -m numato_gpio
Discovered devices:
dev: COM6 | id: 426050343 | ver: 9 | ports: 8 | eol: '\n\r' | iodir: 0xff | iomask: 0x00 | state: 0x00

Yay! That looks better. I assume the 'id' that it shows is the 'id' I use within my python program when opening/creating the device?

clssn, 2021-11-27:
Hi,It's the device ID, stored inside the device as described in the docs: https://numato.com/docs/8-channel-usb-gpio-module-with-analog-inputs/#the-command-set-15Sorry, I only use it on Linux. While the only thing preventing it from being used under windows seems to be the device discovery. If you're interested to contribute, feel free to open a pull request!Best regardsHenning--Henning ClaßenLichtenbergstr. 30/471642 LudwigsburgTel. 07141 1333373Mobil 0173 4233419

clssn, 2022-03-13:
Closing due to inactivity. Feel free to open a new issue relating to this one.

closed 2022-03-13 by clssn