clssn/numato-gpio #14: Some numato devices have uppercase letters in their responses.

trcrenshaw, 2022-03-09:
Came across a board today that had each word capitalized in its response. this board responded with 'Gpio Notify Enabled' but the library expects the response to be 'gpio notify enabled'. Temporary, I've modified the query function to compare the lower case versions by using the str.lowercase() function.

Does anyone know of any cases where this would cause unexpected results. I didn't see anything in the documentation, but I also don't use these a ton so I just want to make sure. I can create a PR for this at some point if it seems like the right way to go.

clssn, 2022-03-10:
Hi, @trcrenshaw ! I originally hoped to handle different behavior by the firmware version, but there are already boards responding with different line endings - even with the same firmware version. So short answer: Yes, I think we need this kind of tweak. Please be so kind and file a PR. Thanks!

clssn, 2023-03-19:
Thanks @lwej !

closed 2023-03-19 by clssn