clssn/numato-gpio #21: Different EOL for notifications and regular responses

zevnikrok, 2022-03-27:
Hi,

I have a Numato GPIO32 board, which returns different EOL sequence for notification (\r\n) compared to regular replies (\n\r).

I was thinking what unpredictable consequences would it make, if we would treat any \r or \n character as a new line character, and then check the extra character if it is \n (after \r) or \r (after \n). and we discard that extra new line character if it is present.

My current workaround is, to set extra EOL property for notifications on startup.

clssn, 2024-03-10:
Hi @zeldi-dev , thanks for your suggestions! Better (too) late than never, I guess ;-) I found the time to re-think the communication with the devices and decided to remove all end-of-line characters from the input. It is actually sufficient to read until the next prompt character, when the exact number of bytes to read is unknown. This simplifies a lot. I plan to release it soon with `0.13.0`. With `0.12.0` numato-gpio already accepts strings as a version from the device and doesn't rely on the version being convertible to an integer. Find the PR with the major changes attached in the development section in the right pane. Hope that helps!

zeldi-dev, 2024-03-10:
Hi. That is even simpler. Great. And thanks for reply!

closed 2024-03-12 by clssn