clssn/numato-gpio #22: Device setup - UnicodeDecodeError

parkourben99, 2022-08-02:
Device: USB GPIO 64
Os: Ubuntu 20.04
Python: 3.8.10
numato-gpio version: 86218b455b0ca25f1d466f89fc1347ecae17739f (latest commit in master) 

```
> python3 -m numato_gpio

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/benjamin/.local/lib/python3.8/site-packages/numato_gpio/__init__.py", line 536, in _poll
    current_value, previous_value, buf = read_notification()
  File "/home/benjamin/.local/lib/python3.8/site-packages/numato_gpio/__init__.py", line 513, in read_notification
    self._q += self._ser.read(1).decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe8 in position 0: unexpected end of data
```

```
> python3 -m numato_gpio.troubleshoot /dev/ttyACM1 64

Testing device /dev/ttyACM1 with pyserial==3.5

writing to /dev/ttyACM1: b'id get\r'
response (19 byte):
b'id get\n\r\xe8Dfp\x92\x96\x99\xa3\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0xe8 0x44 0x66 0x70 0x92 0x96 0x99 0xa3 0xa 0xd 0x3e
--
writing to /dev/ttyACM1: b'ver\r'
response (16 byte):
b'ver\n\rB0000004\n\r>'
0x76 0x65 0x72 0xa 0xd 0x42 0x30 0x30 0x30 0x30 0x30 0x30 0x34 0xa 0xd 0x3e
--
writing to /dev/ttyACM1: b'gpio iomask ffffffffffffffff\r'
response (0 byte):
b''

--
writing to /dev/ttyACM1: b'gpio iomask 0000000000000000\r'
response (0 byte):
b''

--
writing to /dev/ttyACM1: b'gpio readall\r'
response (0 byte):
b''

--
writing to /dev/ttyACM1: b'gpio notify off\r'
response (0 byte):
b''

--
```

```
> screen /dev/ttyACM1

id get
�Dfp����
>ver
B0000004
>gpio readall
FFFFFFFFFEFFFFFF
>gpio notify deisabled
>
```

clssn, 2023-03-19:
First of all, I apologize to reply with such delay.
It looks like your device (or something else) answers with "strange" characters on the `id get` request.
You could try to set a new id e.g. with `id set 00000000` in a screen session. If that doesn't work either, I guess your device's firmware is damaged.

clssn, 2024-03-10:
Closing this stale issue.

closed 2024-03-10 by clssn