clssn/numato-gpio #23: No Devices Found - Numato-GPIO (Python on Mac)

sblynch, 2022-11-12:
Board: Numato USB GPIO 64
OS: MacOS 12.5

When I try to use this code in Python:

my_device_id = 0
gpio.discover()
device = gpio.devices[my_device_id]

I get this exception: 

device = gpio.devices[my_device_id]
KeyError: 0

I've run python3 -m numato_gpio from terminal, and this is what it says:

Discovered devices: (None)

Even though the device is showing up as /dev/tty.usbmodem144101 when I run ls /dev/*usb*

Can you offer guidance on how I can get this USB GPIO 64 to connect in Python on MacOS?  Thanks!

paulvivier, 2022-11-25:
I'm right there with you @sblynch and am trying to solve the same/similar only I have a GPIO 16 device. 
I'm assuming it's not working as part of the "known issues" in the README:
" Only /dev/ACMx devices are scanned which were mapped on the author's Linux " 

Based on that, though, I have a more specific question: 
Are there specific areas of the code that the author would recommend that we look at refactoring so they would also work for devices like **/dev/tty.usbmodem144101** ? My Python knowledge is entry 200 level, but I'd be open to trying. 

BTW, here's my output from running %  python -m numato_gpio (I manually set DEFAULT_DEVICES = ["/dev/tty.usbmodem14301"] )

```
device_file in dev_files: /dev/tty.usbmodem14301
Devices = {}
self._write
self._drain_ser_buffer() = None
Poll thread start: = True
start hasattr
writing id get
Response from device:
id get
00000005
>
eol = 

self.eol = 

start hasattr
writing id get
Response from device:
id get
00000005
>
eol = 

self.eol = 

self._detect_eol() = None
self._MASK_ALL_PORTS = 65535
self._HEX_DIGITS = 4
self._callback = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
self._edge = [None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None]
Discovered devices : (None)
```

clssn, 2024-03-10:
Hi @sblynch and @paulvivier ,  sorry for not being very responsive/helpful recently.
Unfortunately I have zero experience with MacOS. In theory the only OS dependent thing should be the device specifier to initialize the serial connection. I am developing solely in Linux so far and thus I'm only discovering from devices according to Linux conventions (`/dev/ttyACM0` etc.). As @paulvivier seems to already have tried, I'd suggest to adapt the `DEFAULT_DEVICES` array, or just manually instantiate your `NumatoUsbGpio` object when you're using the Python API. Maybe just re-try with the current `0.12.0` which allows for non-integer responses to `ver` queries, or wait for the soon-ready `0.13.0` which drops all end-of-line characters while reading from the device. This makes all weird, but existing combinations of end-of-line sequences from the different device variants acceptable.