clssn/numato-gpio #24: 8 port GPIO not detected (GP80001)

chconnor, 2022-11-24:
Howdy -- having trouble not seeing a device. This is [this unit](https://numato.com/product/8-channel-usb-gpio-module-with-analog-inputs/), with pull-ups.

```
$ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Nov 24 03:46 /dev/ttyACM0
```

Ubuntu 22.04.1 server, updated; python 3.10.6.
numato_gpio version 0.11.0.dev0
user is in dialout group

Here's the troubleshooting output:

```
$ python3 -m numato_gpio.troubleshoot /dev/ttyACM0 8
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000000\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\rA0M10.01\n\r>'
0x76 0x65 0x72 0xa 0xd 0x41 0x30 0x4d 0x31 0x30 0x2e 0x30 0x31 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ff\r'
response (17 byte):
b'gpio iomask ff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00\r'
response (17 byte):
b'gpio iomask 00\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (19 byte):
b'gpio readall\n\rFF\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x46 0xa 0xd 0x3e
```

Any ideas? Thanks!

chconnor, 2022-11-24:
For the record, if I comment out the calls to self.ver, and remove the ver: {} from __str__, things seem to work OK... I can dev.read(x) and it seems to pick up the value correctly.

I'm not getting success trying to use the callback though... maybe you can see a flaw in my test program? I'm not much of a python programmer but I expected this to print out lines whenever the pin goes low/high; I'm using "input" just to keep it from exiting:

```
#!/bin/python3

import numato_gpio as gpio

my_device_id = 0
gpio.discover()
dev = gpio.devices[my_device_id]

dev.setup(7, gpio.IN)

def callback(port, level):
    print("{edge:7s} edge detected on port {port} "
        "-> new logic level is {level}".format(
        edge="Rising" if level else "Falling",
        port=port,
        level="high" if level else "low")
    )

dev.add_event_detect(7, callback, gpio.BOTH)
dev.notify = True

input()
```

chconnor, 2022-11-24:
Ah... the 8 port devices don't have that feature. That's... really annoying. I wish Numato made that clear on the website.

clssn, 2024-03-10:
Hi @chconnor , sorry for not responding for so long. You found it out yourself, the version numbers had to be integers. I fixed this with `v0.12.0`. With `0.13.0` I'll raise errors with clear messages indicating when a device (yes, the 8 port devices) doesn't support notifications, but the user tries to use the "notification API" (`notify` property or `add_event_detect` method). This way, you might have found out quicker that your device doesn't support notifications.

Closing this issue, as the version issue is resolved and the notification issue needs another piece of hardware...

closed 2024-03-10 by clssn