clssn/numato-gpio #25: Unable to Discover Device on Numato 8 Channel USB GPIO Possibly due to Version Number

MitchMagnuson, 2022-12-02:
8 Channel USB GPIO Module
Ubuntu 20.04
Python 3.8.10
numato-gpio 0.11.0.dev0

User is in the dialout group
Works fine with separate 8 channel board with ver '8' firmware
Device is discovered and interactable via putty
[image]
[image]

will-riley, 2022-12-27:
Same problem here.

clssn, 2024-03-10:
Hi @MitchMagnuson and @will-riley , sorry for not responding for such a long time. The issue with versions strings is resolved in `v0.12.0`. Notifications are not supported on the 8 port devices unfortunately. In 'v0.13.0` I'll be raising errors with a clear message to not leave that unnoticed when you try to use the notification related API. Hope that helps!

@will-riley , are you also trying on an 8 port device? If not, may I suggest you open another issue. Maybe my improvements soon released in `v0.13.0` will improve your situation. Otherwise I'd be interested in more details. 

Thanks for reporting, everyone, and sorry again for the lack of responsiveness!

closed 2024-03-10 by clssn

MitchMagnuson, 2024-03-11:
Mitch is no longer with Ainstein. Can you provide some background regarding
this?

*Michael Paulson*
VP of Engineering

785.554.5768
Home: http://ainstein.ai
Shop: https://sensing.ai/