clssn/numato-gpio #28: Numato GPIO not working on HA

AlbertoOcello, 2023-03-20:
HA configuration YAML

numato:
  devices:
    - id: 7
      binary_sensors:
        ports:
          1: 868-1

HA log for numato

2023-03-20 18:39:14.210 INFO (SyncWorker_1) [homeassistant.components.numato] Initializing Numato 32 port USB GPIO expanders with IDs: 
2023-03-20 18:39:21.106 WARNING (MainThread) [homeassistant.setup] Setup of numato is taking over 10 seconds.
2023-03-20 18:39:21.357 INFO (MainThread) [homeassistant.setup] Setup of domain numato took 10.6 seconds
2023-03-20 18:39:22.141 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.numato
2023-03-20 18:39:22.152 ERROR (SyncWorker_3) [homeassistant.components.numato.binary_sensor] Failed to initialize binary sensor '868-1' on Numato device 7 port 1: Device 7 not available

Troubleshooting point by point:
0. OK
1. OK
2.
[  632.286200] usb 1-1.4: new full-speed USB device number 3 using xhci_hcd
[  632.405443] usb 1-1.4: New USB device found, idVendor=2a19, idProduct=0802, bcdDevice= 1.00
[  632.405465] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  632.405473] usb 1-1.4: Product: Numato Lab 32 Channel USB GPIO Module
[  632.405479] usb 1-1.4: Manufacturer: Numato Systems Pvt. Ltd.
[  632.405485] usb 1-1.4: SerialNumber: NLGP210817A0849
[  632.459347] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  632.459525] usbcore: registered new interface driver cdc_acm
[  632.459535] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters´

3.
crw-rw---- 1 root audio 166, 0 Mar 20 14:13 /dev/

I don’t know why the GPIO module is installed with the ownership to audio instead dialout. The first problem i suspect was related to the fact that audio group did not contain root… I fixed it

4.
getent group
audio:x:18
adduser root audio
getent group audio
audio:x:18:root

5.
>id
0000007
>ver
A0000010
>gpio readall
FFFFFFFD
>gpio notify get
Gpio Notify Disabled

6. OK

7.
python3 -m numato_gpio /dev/ttyACM0 32 does not work!! It freeze the terminal ctrl A+ X required

8.
Testing device /dev/ttyACM0 with pyserial==3.5

writing to /dev/ttyACM0: b'id get\r'
response (19 byte):
b'id get\n\r00000007\n\r>'
0x69 0x64 0x20 0x67 0x65 0x74 0xa 0xd 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x37 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'ver\r'
response (16 byte):
b'ver\n\rA0000010\n\r>'
0x76 0x65 0x72 0xa 0xd 0x41 0x30 0x30 0x30 0x30 0x30 0x31 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask ffffffff\r'
response (23 byte):
b'gpio iomask ffffffff\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0x66 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio iomask 00000000\r'
response (23 byte):
b'gpio iomask 00000000\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x69 0x6f 0x6d 0x61 0x73 0x6b 0x20 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio readall\r'
response (25 byte):
b'gpio readall\n\rFFFFFFFD\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x72 0x65 0x61 0x64 0x61 0x6c 0x6c 0xa 0xd 0x46 0x46 0x46 0x46 0x46 0x46 0x46 0x44 0xa 0xd 0x3e
--
writing to /dev/ttyACM0: b'gpio notify off\r'
response (40 byte):
b'gpio notify off\n\rGpio Notify Disabled\n\r>'
0x67 0x70 0x69 0x6f 0x20 0x6e 0x6f 0x74 0x69 0x66 0x79 0x20 0x6f 0x66 0x66 0xa 0xd 0x47 0x70 0x69 0x6f 0x20 0x4e 0x6f 0x74 0x69 0x66 0x79 0x20 0x44 0x69 0x73 0x61 0x62 0x6c 0x65 0x64 0xa 0xd 0x3e
--

AlbertoOcello, 2023-03-24:
Just an update:
Fixed the problem of the version I modified the __init__.py. Now controls is there are characters and remove. 
I also forced to return ver=9 if ver>9. I thought you were not managing ver>9. 
Let me know if is correct
python3 -m numato_gpio didn't work due to a $PATH problem in my installation ( am running HA on a Raspberry with Docker). This need further investigation but I temporary solved by include in the __init__.py code the following:
import os
from pathlib import Path 

# Added to manage Path
newp = "/dev" 
newp = os.chdir( newp )

At the end all troubleshooting step are working as intended but still not initialised by HA.

The only difference I see is the group belonging to /dev/ttyACM0 that in my case is audio while you say always dialout. 
Let me know is the could be the problem...
Thank you again

clssn, 2024-03-10:
Hi @AlbertoOcello , sorry for not being responsive. The device `/dev/ttyACM...` may be dependent on the Linux distribution. I'm using Ubuntu. For using the API I'd suggest you explicitly instantiate a `NumatoUsbGpio` object with the device file path of your choice. I'll probably drop the discovery functionality at some point anyway as it misleads users into thinking their device isn't working.

With respect to your issue with the `ver` I'd ask you to try the `v0.12.0` which allows for any string response when querying the version from the device. This (or depending on my dev velocity `v0.13.0` will be used in Home Assistant core `v2023.3.1`.

I believe your issues should thereby be solved, thus I'm closing this issue. If that's not the case, don't hesitate to re-open or file a new issue!

closed 2024-03-10 by clssn