clssn/numato-gpio #45: Numato Lab 32 Channel USB GPIO not discovered with numato-gpio 0.13.0

jfiger, 2024-03-20:
Still an issue with the latest version on Ubuntu 22.04
```
jfiger@ub2204:~$ lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 2a19:0802 Numato Systems Pvt. Ltd Numato Lab 32 Channel USB GPIO Module
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
jfiger@ub2204:~$ groups jfiger
jfiger : jfiger dialout sudo vboxsf
jfiger@ub2204:~$ pipx install numato-gpio --force
Installing to existing venv 'numato-gpio'
  installed package numato-gpio 0.13.0, installed using Python 3.10.12
  These apps are now globally available
    - numato-discover
  installed package numato-gpio 0.13.0, installed using Python 3.1
done! ✨ 🌟 ✨
jfiger@ub2204:~$ numato-discover
Discovered devices: (None)
jfiger@ub2204:~$ python3 -m numato_gpio
Discovered devices: (None)jfiger@ub2204:~$ python3 -m numato_gpio
jfiger@ub2204:~$ 
```

clssn, 2024-04-17:
Hi @jfiger, maybe ensure first that your device has one of the `/dev/ttyACM` device nodes in the file system, because only devices `/dev/ttyACM0` to `/dev/ttyACM9` are scanned. You can also try `sudo dmesg` right after attaching the device. The kernel log will tell you the device node it assigned .. or an error message if that didn't happen.

I'll try to come up with an update of `numato-discover` which is more verbose.

clssn, 2024-05-23:
I just saw that you installed with pipx but tried to run numato-discover with `python3 -m ` ... This way you started an old version that must already have been installed in your Python environment. I suggest you uninstall that one and make yourself familiar with how to use pipx and what it does. Drop me a note if you need help with that. Pretty sure that the new version can discover your device.

jfiger, 2024-05-23:
You are right. It works !
When this version willl be available in Home Assistant ?

`''
jjfiger@ub2204:~$ sudo ls -al /home/jfiger/.local/bin

total 20

drwxrwxr-x 2 jfiger jfiger 4096 mai   23 10:35 .
drwx------ 6 jfiger jfiger 4096 mars  20 13:38 ..
lrwxrwxrwx 1 jfiger jfiger   68 mai   23 10:35 numato-discover -> /home/jfiger/.local/share/pipx/venvs/numato-gpio/bin/numato-discover
-rwxrwxr-x 1 jfiger jfiger  220 mars  20 13:38 pyserial-miniterm
-rwxrwxr-x 1 jfiger jfiger  222 mars  20 13:38 pyserial-ports

jfiger@ub2204:~$ /home/jfiger/.local/bin/numato-discover

Discovered devices:
numato-gpio v0.13.1a0
dev: /dev/ttyACM0 | id: 268435456 | ver: A0M10.01 | ports: 32 | iodir: 0xffffffff  | iomask: 0x00000000 | state: 0x00000000
jfiger@ub2204:~$
`''

jfiger, 2024-05-23:
I confime it is working OK on HA 2024.5.4

closed 2024-05-23 by jfiger